daliansky/Hackintosh #4: 求助！战神z7-i78172S2下mojave10.14.3的EFI问题！

mayaufo, 2019-03-22:
大佬，我的笔记本是战神Z7-i78172S2，配置是HM87+4720HQ+HD4600+bcm94352z。在安装mojave之后，显卡总是不能正常驱动，所以就想找一个大佬配置好的EFI，但一直没有找到。请大佬帮助找一个，不胜感激！

williambj1, 2019-03-22:
显卡问题可以参考这里[👉🚪](https://blog.daliansky.net/Intel-FB-Patcher-tutorial-and-insertion-pose.html)

mayaufo, 2019-03-22:
> 显卡问题可以参考这里[👉🚪](https://blog.daliansky.net/Intel-FB-Patcher-tutorial-and-insertion-pose.html)

我就是从小兵大佬那里入的坑……除了第一次乱打乱撞，用FB成功驱动，后面就再也没有成功过……并且，当时没有经验，没有备份EFI！！！后来，注册tony，经过向大佬请教，这才让我清楚地认识到，除了同一型号，否则，即便是黑苹果界的大佬也没有办法在看不到真机的情况下搞定所有系统……所以，上来求EFI了……

williambj1, 2019-03-22:
严格按照教程走，不会出问题的，你有成功的经验就说明不需要额外的ssdt之类的东西，确保WEG最新，其他走教程，如果你觉得是Hackintool的问题，你可以看[WEG的官方文档（英文）](https://github.com/acidanthera/WhateverGreen/blob/master/Manual/FAQ.IntelHD.en.md)，或者过几天看PR的中文版。

mayaufo, 2019-03-22:
> 严格按照教程走，不会出问题的，你有成功的经验就说明不需要额外的ssdt之类的东西，确保WEG最新，其他走教程，如果你觉得是Hackintool的问题，你可以看[WEG的官方文档（英文）](https://github.com/acidanthera/WhateverGreen/blob/master/Manual/FAQ.IntelHD.en.md)，或者过几天看PR的中文版。
> 大佬也是自己学出来的，依赖他人EFI并不靠谱。

谢谢指导！我按官方文档再重新尝试。

closed 2019-03-23 by daliansky