daliansky/Hackintosh #29: 跪求联想拯救者15ikbn-i7-7700HQ 的10.14.5  EFI

haohaoxueba, 2019-06-18:
各位大神:
    你们好,我希望能得到联想拯救者15ikbn-i7-7700HQ 的10.14.5  EFI,由于 HDMI 和 typec 的问题,已经折磨了我很久很久,我的显卡是 Gtx1050 ,独显 HD-630  显卡  英特尔 HD Graphics 630,希望能解决严重的 HDMI 和 typec 问题,小弟在此先谢过各位大佬了
[配置表.txt](https://github.com/daliansky/Hackintosh/files/3301359/default.txt)

haohaoxueba, 2019-06-18:
--------[ 概览 ]----------------------------------------------------------------------------------

  电脑型号            联想 80WW 笔记本电脑
  操作系统            Windows 10 专业版 64位 ( DirectX 12 )

  处理器              英特尔 Core i7-7700HQ @ 2.80GHz 四核
  主板                联想 Provence-5R3 ( 100 Series/C230 Series 芯片组 Family - A152 )
  显卡                Nvidia GeForce GTX 1050 ( 2 GB / 联想 )
  内存                16 GB ( 三星 )
  主硬盘              英特尔 SSDPEKKW512G8 ( SSD7代/512 GB / 固态硬盘 )
  显示器              友达 AUO61ED ( 15.5 英寸  )
  声卡                瑞昱  @ 英特尔 High Definition Audio 控制器
  无线网卡            Atheros Qualcomm Atheros QCA9377 Wireless Network Adapter / 联想

--------[ 主板 ]----------------------------------------------------------------------------------
  主板型号            联想 Provence-5R3
  芯片组              100 Series/C230 Series 芯片组 Family - A152
  BIOS版本            4KCN43WW  /  BIOS程序发布日期: 04/12/2018
--------[ 处理器 ]--------------------------------------------------------------------------------
  处理器              英特尔 Core i7-7700HQ @ 2.80GHz 四核
  速度                2.81 GHz (100 MHz x 28.0)
  处理器数量          核心数: 4 / 线程数: 8
  插槽/插座           U3E1
  一级数据缓存        4 x 32 KB, 8-Way, 64 byte lines
  一级代码缓存        4 x 32 KB, 8-Way, 64 byte lines
  二级缓存            4 x 256 KB, 4-Way, 64 byte lines
  三级缓存            6 MB, 12-Way, 64 byte lines
  特征                MMX, SSE, SSE2, SSE3, SSSE3, SSE4.1, SSE4.2, HTT, EM64T, EIST, Turbo Boost
--------[ 硬盘 ]----------------------------------------------------------------------------------

  产品                英特尔  SSDPEKKW512G8 (固态硬盘)
  大小                512 GB
  产品                希捷 ST1000LM035-1RK172
  大小                1 TB
--------[ 显卡 ]----------------------------------------------------------------------------------

  显卡                英特尔 HD Graphics 630
  显存                128 MB
  制造商              联想
  制造商              Intel
  驱动版本            23.20.16.4973
  驱动日期            20180228

  主显卡              Nvidia GeForce GTX 1050
  显存                2 GB
  制造商              联想
  制造商              Nvidia
  驱动版本            26.21.14.3039
  驱动日期            20190417

--------[ 其他设备 ]------------------------------------------------------------------------------

  无线网卡            Atheros Qualcomm Atheros QCA9377 Wireless Network Adapter
  制造商              联想

  网卡                瑞昱 RTL8168/8111/8112 Gigabit Ethernet Controller
  制造商              联想

  声卡                瑞昱  @ 英特尔 High Definition Audio 控制器

  键盘                PS/2 标准键盘
  键盘                HID 标准键盘
  鼠标                Synaptics 指点杆
  鼠标                HID-compliant 鼠标

GG998667, 2019-07-14:
哈哈哈跟我一样，我差弄了一个感觉还行

Jgump, 2019-07-23:
直接在本站搜索，只要是主板，跟cpu型号对得起基本没问题

haohaoxueba, 2019-08-05:
> 哈哈哈跟我一样，我差弄了一个感觉还行

你的hdmi可以用吗

haohaoxueba, 2019-08-12:
> 直接在本站搜索，只要是主板，跟cpu型号对得起基本没问题

我装上了,我是要用HDMI和type-c分屏

lxfwelldone, 2019-09-26:
同联想拯救者，不过我的是1+128的，也想装个黑苹果，兄弟，有教程和资源吗？

Jgump, 2019-09-26:
按cpu型号路找efi试一下

发自我的iPhone

haohaoxueba, 2019-09-26:
> 同联想拯救者，不过我的是1 + 128的，也想装个黑苹果，兄弟，有教程和资源吗？

ok,这是我的EFI,我已经调试好了,TypeC接口可以连接屏幕使用,安装教程找小兵
http://blog.dundunduang.com/EFI.zip

closed 2020-11-04 by daliansky