daliansky/Hackintosh #120: 索尼笔记本SVF14326SCB EFI

luobo180180, 2019-12-25:
索尼笔记本SVF14326SCB EFI

黑苹果10.14.6驱动以下设备
集成显卡
声卡
有线网卡
鼠标键盘

由于文件大小限制，删除了一些主题，WINDOWS启动引导
[SVF14326SCB EFI.zip](https://github.com/daliansky/Hackintosh/files/4000458/SVF14326SCB.EFI.zip)

daliansky, 2019-12-26:
> 索尼笔记本SVF14326SCB EFI
> 
> 黑苹果10.14.6驱动以下设备
> 集成显卡
> 声卡
> 有线网卡
> 鼠标键盘
> 
> 由于文件大小限制，删除了一些主题，WINDOWS启动引导
> [SVF14326SCB EFI.zip](https://github.com/daliansky/Hackintosh/files/4000458/SVF14326SCB.EFI.zip)

建议自己建立一个仓库，然后PR给我

hclike, 2020-01-04:
> 索尼笔记本SVF14326SCB EFI
> 
> 黑苹果10.14.6驱动以下设备
> 集成显卡
> 声卡
> 有线网卡
> 鼠标键盘
> 
> 由于文件大小限制，删除了一些主题，WINDOWS启动引导
> [SVF14326SCB EFI.zip](https://github.com/daliansky/Hackintosh/files/4000458/SVF14326SCB.EFI.zip)
可以交流下么，我的是索尼svf143a1qt型号，可以用你的吗

luobo180180, 2020-01-04:
可以啊，先试了再说，我都是网上到处下载试的，最后就可以了

hclike, 2020-01-04:
你qq多少我加下你------------------ 原始邮件 ------------------

 
 
 
 
 ------------------ 原始邮&#x4EF6; ------------------ 
 发&#x4EF6;人: "Hclike"<notifications@github.com&gt;;  
 收&#x4EF6;人: "daliansky/Hackintosh"<Hackintosh@noreply.github.com&gt;;  
 
 
 
   
   
   
 由于文&#x4EF6;大小限制，删除了一些主题，WINDOWS启动引导 
    
&mdash;

luobo180180, 2020-01-04:
QQ328790763,验证全部1

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "Hclike"<notifications@github.com&gt;; 
发送时间: 2020年1月4日(星期六) 晚上9:10
收件人: "daliansky/Hackintosh"<Hackintosh@noreply.github.com&gt;; 
抄送: "び小胡在线"<luobo180@qq.com&gt;; "Author"<author@noreply.github.com&gt;; 
主题: Re: [daliansky/Hackintosh] 索尼笔记本SVF14326SCB EFI (#120)

你qq多少我加下你------------------ 原始邮件 ------------------ 
 发件人: "luobo180180"<notifications@github.com&gt; 
 发送时间: 2020年1月4日(星期六) 晚上8:59 
 收件人: "daliansky/Hackintosh"<Hackintosh@noreply.github.com&gt;; 
 抄送: "Hclike"<2801849952@qq.com&gt;;"Comment"<comment@noreply.github.com&gt;; 
 主题: Re: [daliansky/Hackintosh] 索尼笔记本SVF14326SCB EFI (#120) 
 
 
 可以啊，先试了再说，我都是网上到处下载试的，最后就可以了  
   
   
   
   
  ------------------&nbsp;原始邮件&nbsp;------------------  
  发件人: "Hclike"<;;   s@github.com＆gt通知
  发送时间: 2020年1月4日(星期六) 上午10:32  
  收&amp;#x4EF6;人: "daliansky/Hackintosh"<Hackintosh@noreply.github.com&amp;gt;;   
  抄送: "び小胡在线"<luobo180@qq.com&amp;gt;; "Author"<author@noreply.github.com&amp;gt;;   
  主题: Re: [daliansky/Hackintosh] 索尼笔记本SVF14326SCB EFI (#120)  
   
   
   
     
  索尼笔记本SVF14326SCB EFI  
     
  黑苹果10.14.6驱动以下设备  
   集成显卡  
   声卡  
   有线网卡  
   鼠标键盘  
     
  由于文&amp;#x4EF6;大小限制，删除了一些主题，WINDOWS启动引导  
   SVF14326SCB EFI.zip  
   可以交流下么，我的是索尼svf143a1qt型号，可以用你的吗  
      
  —  
  You are receiving this because you authored the thread.  
  直接回复此电子邮件，在GitHub上查看或取消订阅。  
 ＆mdash; 
 您收到此邮件是因为您发表了评论。 
 直接回复此电子邮件，在GitHub上查看或取消订阅。 
— 
您收到此消息是因为您创建了线程。
直接回复此电子邮件，在GitHub上查看，或取消订阅。

luobo180180, 2020-01-04:
完整版的

链接：https://pan.baidu.com/s/1erazUn8YZqTCptK_u3QoNA&nbsp;
提取码：scns&nbsp;
复制这段内容后打开百度网盘手机App，操作更方便哦

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "daliansky"<notifications@github.com&gt;; 
发送时间: 2019年12月26日(星期四) 晚上9:59
收件人: "daliansky/Hackintosh"<Hackintosh@noreply.github.com&gt;; 
抄送: "び小胡在线"<luobo180@qq.com&gt;; "Author"<author@noreply.github.com&gt;; 
主题: Re: [daliansky/Hackintosh] 索尼笔记本SVF14326SCB EFI (#120)

 
索尼笔记本SVF14326SCB EFI
 
黑苹果10.14.6驱动以下设备
 集成显卡
 声卡
 有线网卡
 鼠标键盘
 
由于文件大小限制，删除了一些主题，WINDOWS启动引导
 SVF14326SCB EFI.zip
  
建议自己建立一个仓库，然后PR给我
 
— 
您收到此消息是因为您创建了线程。
直接回复此电子邮件，在GitHub上查看，或取消订阅。

hclike, 2020-01-04:
提取码乱码了……

发自我的iPhone

 
 
 提取码：scns&amp;nbsp; 
 
 
 
 
 ------------------&amp;nbsp;原始邮件&amp;nbsp;------------------

luobo180180, 2020-01-04:
完整版的EFI百度盘下载

链接：https://pan.baidu.com/s/1erazUn8YZqTCptK_u3QoNA&nbsp;
提取码：scns&nbsp;
复制这段内容后打开百度网盘手机App，操作更方便哦

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "Hclike"<notifications@github.com&gt;; 
发送时间: 2020年1月4日(星期六) 晚上9:10
收件人: "daliansky/Hackintosh"<Hackintosh@noreply.github.com&gt;; 
抄送: "び小胡在线"<luobo180@qq.com&gt;; "Author"<author@noreply.github.com&gt;; 
主题: Re: [daliansky/Hackintosh] 索尼笔记本SVF14326SCB EFI (#120)

你qq多少我加下你------------------ 原始邮件 ------------------ 
 发件人: "luobo180180"<notifications@github.com&gt; 
 发送时间: 2020年1月4日(星期六) 晚上8:59 
 收件人: "daliansky/Hackintosh"<Hackintosh@noreply.github.com&gt;; 
 抄送: "Hclike"<2801849952@qq.com&gt;;"Comment"<comment@noreply.github.com&gt;; 
 主题: Re: [daliansky/Hackintosh] 索尼笔记本SVF14326SCB EFI (#120) 
 
 
 可以啊，先试了再说，我都是网上到处下载试的，最后就可以了  
   
   
   
   
  ------------------&nbsp;原始邮件&nbsp;------------------  
  发件人: "Hclike"<;;   s@github.com＆gt通知
  发送时间: 2020年1月4日(星期六) 上午10:32  
  收件人: "daliansky/Hackintosh"<Hackintosh@noreply.github.com&gt;;   
  抄送: "び小胡在线"<luobo180@qq.com&gt;; "Author"<author@noreply.github.com&gt;;   
  主题: Re: [daliansky/Hackintosh] 索尼笔记本SVF14326SCB EFI (#120)  
   
   
   
     
  索尼笔记本SVF14326SCB EFI  
     
  黑苹果10.14.6驱动以下设备  
   集成显卡  
   声卡  
   有线网卡  
   鼠标键盘  
     
  由于文件大小限制，删除了一些主题，WINDOWS启动引导  
   SVF14326SCB EFI.zip  
   可以交流下么，我的是索尼svf143a1qt型号，可以用你的吗  
      
  —  
  You are receiving this because you authored the thread.  
  直接回复此电子邮件，在GitHub上查看或取消订阅。  
 ＆mdash; 
 您收到此邮件是因为您发表了评论。 
 直接回复此电子邮件，在GitHub上查看或取消订阅。 
— 
您收到此消息是因为您创建了线程。
直接回复此电子邮件，在GitHub上查看，或取消订阅。

hclike, 2020-01-04:
加你了，你通过一下，2801849952------------------ 原始邮件 ------------------

 
 
 
 
 
 
 ------------------ 原始邮&#x4EF6; ------------------ 
 
 
 
 你qq多少我加下你------------------ 原始邮&#x4EF6; ------------------  
  发&#x4EF6;人: "luobo180180"<notifications@github.com&gt;  
   
   
     
     
     
     
   ------------------ 原始邮&#x4EF6; ------------------   
   发&#x4EF6;人: "Hclike"<;;   s@github.com＆gt通知 
     
     
     
       
       
       
        
   直接回复此电子邮&#x4EF6;，在GitHub上查看或取消订阅。   
  ＆mdash;  
  您收到此邮&#x4EF6;是因为您发表了评论。  
  直接回复此电子邮&#x4EF6;，在GitHub上查看或取消订阅。  
 您收到此消息是因为您创建了线程。 
 直接回复此电子邮&#x4EF6;，在GitHub上查看，或取消订阅。

closed 2020-11-04 by daliansky