daliansky/Hackintosh #122: 雷神G7000S的EFI

liang0911, 2019-12-28:
码云仓库：https://gitee.com/DYYX19980911/Thunderobot-G7000S-Hackintosh-By-Clover
GitHub仓库：https://github.com/liang0911/Thunderobot-G7000S-Hackintosh-By-Clover

athlonreg, 2020-02-05:
> 码云仓库：https://gitee.com/DYYX19980911/Thunderobot-G7000S-Hackintosh-By-Clover
> GitHub仓库：https://github.com/liang0911/Thunderobot-G7000S-Hackintosh-By-Clover

大致看了一下，还有不少可以完善的地方，efi里面的origin是否是本机的原始acpi？能否提供一下设备管理器的人体输入学设备截图呢。我试着看看能不能帮忙完善一下
点击链接加入群聊【远景雷神黑苹果交流群】：https://jq.qq.com/?_wv=1027&k=5iU5ImF
可以加一下群聊大家一块探讨

liang0911, 2020-02-05:
> 大致看了一下，还有不少可以完善的地方，efi里面的origin是否是本机的原始acpi？能否提供一下设备管理器的人体输入学设备截图呢。我试着看看能不能帮忙完善一下
> 点击链接加入群聊【远景雷神黑苹果交流群】：https://jq.qq.com/?_wv=1027&k=5iU5ImF
> 可以加一下群聊大家一块探讨

是本机的原始acpi，通过Clover+F4获取的，亮度补丁是后面加进去的。因为没有什么时间，我最近已经没有在搞黑苹果了，维护也有一段时间没有更新了。

closed 2020-11-04 by daliansky