daliansky/Hackintosh #126: i7-10710u

acoutts, 2020-01-02:
Hi - is there any success yet on a 10th gen i7?

Am trying to get Lenovo X1C7 to boot (6 core i7-10710u) but no luck so far.

My clover / OC:
https://github.com/acoutts/x1c7-hackintosh

daliansky, 2020-01-09:
https://github.com/daliansky/XiaoXinPro-13-2019-hackintosh
[image]

osxtremez, 2020-01-28:
Hello,
I'm also trying to install Mojave on i7-10710U (Dell XPS13-7390.
Followed you link above but was not successful.
here's the screenshot.
pls advise.
TIA.
[image]

evgenyrud, 2020-02-06:
Are there good new with i7-10710u?
Created Clover EFS but not able to boot Catalina USB. Stuck from the very beginning.

astrocycles, 2020-04-07:
Hi,
I'm new to this.
Should the CPU be exactly the same as above or this could work with NUC 10 for example ? 
many thanks

1alessandro1, 2020-05-04:
Hi, any success with IceLake? Particularly with Lenovo IceLake laptops?

Ali-Jamali, 2020-06-03:
> Hi, any success with IceLake? Particularly with Lenovo IceLake laptops?

I try it in 10.15.5 with clover and OC 
Both not booting

griffins, 2020-06-21:
i managed to boot on open core, the camera is working, Bluetooth, keyboard, slight issues with the touchpad,   am supposed to replace the wifi card but am yet, i have no igpu acceleration yet.

griffins, 2020-06-23:
Touch Pad was fixed.  My IGPU isnt for ice lake but some sort of comet lake, device id 0x9b41, i believe some 630 UHD graphics should work,

RotatingFans, 2020-06-23:
@griffins Is that with the 10710?

griffins, 2020-06-23:
yep!

RotatingFans, 2020-06-23:
Be sure to keep us updated! It seems you may be one of the first.

griffins, 2020-06-23:
i will

LEXUGE, 2020-06-27:
> i managed to boot on open core, the camera is working, Bluetooth, keyboard, slight issues with the touchpad, am supposed to replace the wifi card but am yet, i have no igpu acceleration yet.
> 
> its a Dell XPS13-7390
> PS, i swapped the nvme to a samsung evo pro.

Would you like to upload your EFI to a GitHub repo?

griffins, 2020-06-27:
https://github.com/griffins/dell-xps-7390

Any help to sort the GPU acceleration is appreciated.

griffins, 2020-06-27:
according to this we have a uhd 620 comet lake version 
https://www.notebookcheck.net/Intel-Core-i7-10510U-Laptop-Processor-Comet-Lake-U.430586.0.html

griffins, 2020-06-27:
faking the device id gets GPU acceleration but it get stuck at the ioconsoleusers gioscreenlockstate 3 just before login screen. so its a graphics issue. the only way to boot is to use a none-existent device id and let the os fall back to using builtin graphics.

macblizzard, 2020-07-19:
> faking the device id gets GPU acceleration but it get stuck at the ioconsoleusers gioscreenlockstate 3 just before login screen. so its a graphics issue. the only way to boot is to use a none-existent device id and let the os fall back to using builtin graphics.

to which older version did you fake it to? did you follow any guide to fake it?

closed 2020-11-04 by daliansky