daliansky/Hackintosh #170: surface pro7

fffoobibi, 2020-04-08:
你们右pro7的efi么,谢谢

g707175425, 2020-06-26:
同求

EngLearnsh, 2020-06-27:
[https://github.com/badstorm/surface-pro-7-opencore](url)
问题一堆，基本无法使用

ati46, 2020-08-10:
pro7的efi
链接: https://pan.baidu.com/s/1Jr6lH91t6DsGwfdtuohC-g 提取码: tn3f
1. usb和sd卡正常
2.蓝牙，wifi正常。使用论坛大神的intel201驱动
3.surface type cover正常
4.触摸板手势正常
5.hidpi正常

需要解决的问题：
1.显卡无法驱动，显示19M
2.电池无法显示，应该是没驱动
     1.合盖无法息屏
     2.进入睡眠后界面会卡在睡眠前的画面，敲击键盘后可以正常使用
3.虽然触摸板手势可用，但是设置界面触摸板选项点进入，显示无法连接到触摸板，导致无法设置轻触点击
目前我在用的efi，  剩余问题希望有大佬可以一起解决

[image]

g707175425, 2020-09-25:
@ati46 多谢大佬

g707175425, 2020-09-25:
可以试试这个 驱动显卡了
https://github.com/ati46/surfacepro7-oc

closed 2020-11-04 by daliansky